kraiskil/onnx2c #6: ModelProto::ParseFromIstream missing

mrzealot, 2021-01-24:
Hi! I'm trying to compile this repo, but get the following message when running `make`:

```
src/main.cc:34:13: error: ‘class onnx::ModelProto’ has no member named ‘ParseFromIstream’; did you mean ‘ParseFromString’?
```

I'm not really familiar with ProtoBuf stuff (or ONNX, for that matter), but I did everything like the readme suggests, dependencies installed, submodules at their respective commits. Any idea what I'm missing?

kraiskil, 2021-01-27:
Sounds like a ProtoBuf version mismatch. I'm not too familiar with it either - it is just a ONNX requirement.
Which OS are you on, and which version of ProtoBuf do you have installed? Recently tested are Ubuntu 16.04, 20.10 and MacOS BigSur - all with their "default" versions of ProtoBuf.

kraiskil, 2021-01-27:
I wrote too hastily. Master branch was indeed broken on 0186da3535e37cefe8448d4c065fdcd035a44816 on Ubuntu 16.04 (when building clean - which I didn't, so I didn't notice it was broken), and I could repeat the problem you saw. 
The fix is to modify `onnx/onnx/onnx.proto` (remove the last line), and rebuild. I added a note on this into the `readme.md`. Hope this fixes it for you too.

mrzealot, 2021-01-27:
Excellent, this fixes it indeed! Thanks a bunch!

closed 2021-01-27 by mrzealot

mrzealot, 2021-01-27:
Just for the sake of completeness, `lsb_release -a` says:

```
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal
```

`apt` says:

```
libprotobuf-dev = (3.6.1.3-2ubuntu5).
protobuf-compiler = (3.6.1.3-2ubuntu5).
```

`cmake` says:

```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /path/to/onnx2c/build
```

So whatever change made it work out of the box should have happened between Ubuntu 20.04 and 20.10...