kraiskil/onnx2c #7: Convolution bias bug

mrzealot, 2021-03-05:
Hi! Was tinkering with the converter and found a bug on [this line](https://github.com/kraiskil/onnx2c/blob/a7b27e48a3dabb6e68933eb5652cfa4f6b16d6e1/src/nodes/conv.h#L222). Seems like the bias addition is indexed along the incorrect dimension (I think it should be `[m]` instead of `[b]`). Just a heads up...

kraiskil, 2021-03-05:
Yeah, that does indeed look like it should be `[m]`. Would you happen to have a nice small test case to trigger this?

mrzealot, 2021-03-05:
Nah, sorry, just a huge (thereby not very helpful, and unfortunately also non-public) input, after which I was just glad to have found the problem. But any mini example with a multi-map kernel will probably do...

kraiskil, 2021-03-06:
Fixed now, thanks for the detective work!

closed 2021-03-06 by kraiskil