kraiskil/onnx2c #10: Build fails on macOS

morungos, 2021-11-27:
I'm now using a Monterey Mac - macOS 12.0.1, but the build fails. It seems that `-Wall` now defaults to throwing errors on `overrides a member function but is not marked 'override'`. I've attached a log of the `make` output: [log.txt](https://github.com/kraiskil/onnx2c/files/7612222/log.txt).

gcc reports a version:

```
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.29.3)
Target: x86_64-apple-darwin21.1.0
```

Simple resolution: I added `-Wno-inconsistent-missing-override` to the `CMAKE_C_FLAGS` and `CMAKE_CXX_FLAGS` values in `CMakeLists.txt`

kraiskil, 2021-11-28:
Thanks for the heads up. Trust Apple to be on top of trends... or is it just me who should finally embrace C++11?

I think the proper solution would be to add the `override` keywords, but the lack of them might spread all over the place, so deprecating those warnings sounds like a better idea, at least as a stop-gap.

I can't reproduce this problem for testing, but is the flag needed also for `CMAKE_C_FLAGS`, not only `CMAKE_CXX_FLAGS`? If so, this means somewhere in onnx2c there is a mixup with which flags to use.

kraiskil, 2022-02-17:
The above commit adds `override` to all (I hope) overriding functions, which I think fixes this issue. Please re-open if I missed some.

closed 2022-02-17 by kraiskil