kraiskil/onnx2c #16: Concat's input tensors must have the same shape, except for the dimension size of the axis to concatenate on.

CarlosPurchase-Galarza, 2022-04-04:
I get the following error when trying to compile the given .onnx file.

```
2022-04-04 17-52-38.409 [Fatal] (resolveOutput) Concat's input tensors must have the same shape, except for the dimension size of the axis to concatenate on.
onnx2c: /home/carlos/onnx2c/src/nodes/concat.h:109: virtual void toC::Concat::resolveOutput(const std::vector<const toC::Tensor*>&, std::vector<toC::Tensor*>&): Assertion `false' failed.
Aborted (core dumped)
```
[lenet_mnist.zip](https://github.com/kraiskil/onnx2c/files/8411691/lenet_mnist.zip)

Thanks

CarlosPurchase-Galarza, 2022-04-04:
Issue has been resolved by replacing the first line with the second line in the PyTorch model.

```
#y = y.view(y.shape[0], -1)
y = y.reshape(-1,256)
```

closed 2022-04-04 by CarlosPurchase-Galarza

kraiskil, 2022-04-05:
Hi,
good thing you got it resolved. I'm not quite sure I understood: is the fix you mentioned a work-around onnx2c being buggy, or was the generated network just bad before that fix?

If it still is a bug with onnx2c, then usually it is an earlier node that resolves its inputs incorrectly, and this wrong size is propagated possibly over several nodes before getting caught by a later node - in this case by Concat.