kraiskil/onnx2c #18: Resize operator

lingxutian, 2022-06-01:
[Fatal] (resolve) Unimplemented: Resize 'sizes' input is not a compile-time constant
onnx2c: /home/zentech/onnx2c/src/nodes/resize.h:82: virtual void toC::Resize::resolve(): Assertion `false' failed.
How should I modify it?

kraiskil, 2022-06-02:
Hi lingxutian, 
this error means that your graph has a data dimension that is defined at run time (i.e. the input data defines a tensor size). This is valid ONNX, but not supported by onnx2c, since it mean dynamic memory handling - a many times problematic thing on microcontrollers.

If your ONNX graph design requires dynamic inputs sizes, then there is currently not much that can be done. Having onnx2c support that would require deep changes in its implementation. 
One option is to modify your graph to not need dynamic input. E.g. if it is with images, scale the image to a fixed size before passing it to the graph for processing.

But these run time defined sizes can also happen by accident, since higher level frameworks don't really see dynamic sizes as problematic. I've noticed sometimes running the onnx file through e.g. onnxsim swaps out a dynamic tensor size for a fixed one.

Hope this helps :)

PasaOpasen, 2022-12-25:
@kraiskil do u have some extended onnx translator which supports dynamic features?