kraiskil/onnx2c #23: CMake fails

polluks, 2022-10-28:
```
$ cmake ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.20.1")
error: unknown option `show-current'

kraiskil, 2022-10-28:
This is git failing. `git branch --show-current` was introduced in git 2.22 (e.g. https://stackoverflow.com/questions/1417957/show-just-the-current-branch-in-git/55088865#55088865 ). 
The problem is in the cmake timestamp subrepo (https://github.com/kraiskil/cmake_timestamp) not handling older versions of git cleanly.
Since git 2.22 is only a few years old, it would be nice if this problem could be worked around, especially since recording the build info into onnx2c isn't that important a feature in the first place.

kraiskil, 2022-10-28:
Oh, and a quick work-around before a proper fix is available would be to comment out line 52:
`git_cmd( "branch --show-current" GIT_BRANCH)` in file `cmake_timestamp/CMakeLists.txt`

kraiskil, 2022-10-30:
Latest master should now compile without warnings. Though also without branch info in the version string.
Please re-open the issue if I goofed something.

Thanks for reporting!

closed 2022-10-30 by kraiskil