kraiskil/onnx2c #26: onnx2c/src/nodes/cast.cc:36: virtual void toC::Cast::resolve(): Assertion `false' failed.

PasaOpasen, 2022-12-27:
Tried to compile this model https://dropmefiles.com/MkdyS

with structure 
[image]

what can be wrong?

kraiskil, 2022-12-28:
What is happening here is, as you write in the title of this issue, you are hitting [this](https://github.com/kraiskil/onnx2c/blob/87a548021c4da69ed7bc0ce4c7c41eaae0249282/src/nodes/cast.cc#L36) error. 

This is an unimplemented feature in onnx2c. 
I don't see the full output of onnx2c, but if the  error message linked to above is not printed out, then it is also an error in onnx2c.

The entire concept of a cast operation should be folded into the output of the predecessor node of the cast, as having a separate node just means a redundant copy of data. On the other hand, the C compiler should maybe have such an optimization pass?

PasaOpasen, 2022-12-29:
```
╭─ pasa  ~/Job/__tmp/onnx2c/build  📁 4 🗎 9 🔗 0   master 
╰> $ ./onnx2c ../../craft_fix_input.onnx > model.c
2022-12-29 21-07-35.866 [Fatal] (resolve) Unimplemented casting to requested type
onnx2c: /home/pasa/Job/__tmp/onnx2c/src/nodes/cast.cc:36: virtual void toC::Cast::resolve(): Assertion `false' failed.
```

its whole stack

shubham0204, 2024-02-09:
@kraiskil is there any update to this error? 

```
2024-02-09 09-56-10.197 [Info] (onnx_ir_version) Model has multiple opset versions.
2024-02-09 09-56-10.283 [Info] (resolve) 'Cast' node found.
2024-02-09 09-56-10.284 [Info] (resolve) 'Cast' node found.
2024-02-09 09-56-10.284 [Info] (resolve) 'Cast' node found.
2024-02-09 09-56-10.284 [Info] (resolve) 'Cast' node found.
2024-02-09 09-56-10.284 [Warn] (Elementwise) Round operand implementation is not strictly conformant
2024-02-09 09-56-10.285 [Info] (resolve) 'Cast' node found.
2024-02-09 09-56-10.285 [Fatal] (resolve) Unimplemented casting to requested type
onnx2c: ~/onnx2c/src/nodes/cast.cc:34: virtual void toC::Cast::resolve(): Assertion `false' failed.
```

kraiskil, 2024-02-11:
@shubham0204 sorry, no. 
I've been working in the background for the possibility for optimizing input onnx networks (folding Casts away being one of them), so I didn't pay much attention to improving Cast.

Looking at the code at `src/nodes/cast.cc:34`, it should be fairly straight-forward to extend to support other datatypes. Unfortunately onnx doesn't seem to have any backend test for integer casting, so I am reluctant to add such code "blindly". I guess it could be added even without a unittest as long as someone has tested the patch to be working...

kraiskil, 2024-05-19:
Might be fixed now, please re-open new issue, if not.

closed 2024-05-19 by kraiskil