kraiskil/onnx2c #33: make: *** No rule to make target 'onnx2c'.  Stop.

hakam00ra, 2023-09-01:
I'm trying to build onnx2c but have been stuck on the make part. I've installed protobuf via vcpkg and this is what I get when 
I try to build: 

Using cmake -DCMAKE_BUILD_TYPE=Release .. gave me these errors:

> -- Compiling and running to test HAVE_STD_REGEX
    -- Performing Test HAVE_STD_REGEX -- failed to compile
    -- Compiling and running to test HAVE_GNU_POSIX_REGEX
    -- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
    -- Compiling and running to test HAVE_POSIX_REGEX
    -- Performing Test HAVE_POSIX_REGEX -- failed to compile
CMake Error at benchmark/CMakeLists.txt:308 (message):

So I changed it to this:

> PS C:\Users\Haris\Desktop\python\onnx2x\onnx2c\build> cmake -DCMAKE_CROSSCOMPILING=1 -DRUN_HAVE_STD_REGEX=0 -DRUN_HAVE_POSIX_REGEX=0 ..
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19045.
-- git version: v1.7.0-10-g77d1e74d normalized to 1.7.0.10
-- Version: 1.7.0.10
-- Performing Test HAVE_STD_REGEX -- success
-- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
-- Performing Test HAVE_POSIX_REGEX -- success
-- Compiling and running to test HAVE_STEADY_CLOCK
-- Performing Test HAVE_STEADY_CLOCK -- failed to compile
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE
-- Configuring done (3.0s)
-- Generating done (3.0s)
-- Build files have been written to: C:/Users/Haris/Desktop/python/onnx2x/onnx2c/build

And then: 

> PS C:\Users\Haris\Desktop\python\onnx2x\onnx2c\build> make onnx2c
    make: *** No rule to make target 'onnx2c'.  Stop.

protoc --version returns libprotoc 3.21.12

Any ideas or anything else I should add?

thanks

kraiskil, 2023-09-01:
As far as I know, this is the first attempt to build onnx2c on Windows. So a bumpy ride is expected. 

In stead of running `make`, would a build with CMake work? `cmake --build . --config Release`?

hakam00ra, 2023-09-01:
I get this with your suggestion:

>  1>Checking Build System
  -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19045.
  -- git version: v1.7.0-10-g77d1e74d normalized to 1.7.0.10
  -- Version: 1.7.0.10
  -- Performing Test HAVE_STD_REGEX -- success
  -- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
  -- Performing Test HAVE_POSIX_REGEX -- success
  -- Performing Test HAVE_STEADY_CLOCK -- failed to compile
  -- Configuring done (1.4s)
  -- Generating done (17.7s)
  -- Build files have been written to: C:/Users/Haris/Desktop/python/onnx2x/onnx2c/build
  cl : command line error D8021: invalid numeric argument '/Werror' [C:\Users\Haris\Desktop\python\onnx2x\onnx2c\build\cm
  ake_timestamp\timestamp.vcxproj] 

I'll keep trying and post a possible solution here although I guess I'll probably just use a linux systrem.

thanks

kraiskil, 2023-09-01:
Right... Seems some gcc/clang command line options are hard-coded into the CMake files. A fix for these is of course welcome, but there might be quite a lot of spots to fix. Linux is for sure the easier route to take for now.

kraiskil, 2024-05-29:
Some new Windows build related discussion in #49, closing this in favour of moving the discussion there.

closed 2024-05-29 by kraiskil