kraiskil/onnx2c #34: fatal error: 'parseAttributes' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]

YJack0000, 2023-09-13:
When compiling with c++ >= 11 will have this override error

YJack0000, 2023-09-13:
#35

kraiskil, 2023-09-13:
Interesting. The top level CMakeListst.txt does specify C++11.
I've compiled with both GCC and Clang, and don't see this warning/error. Which compiler are you using?

YJack0000, 2023-09-13:
Sorry for the mis understanding of the problem. 
I initially thought it is a CXX version problem because I saw someone on the Internet saying that inconsistent-missing-override is a new thing in C++11.

After trying on different devices, I found that it can compiled successfully in C++11 on my linux laptop. 

So, I think the problem might be from xcode, because the compiler that I'm using to get that error is the one in the image below. 
[image]

And the error log is 
[image]

kraiskil, 2023-09-13:
Ok, that clang looks fairly recent. I've tried with Clang 10 or so, if memory serves.

The main problem is the -Werror flag that converts even fairly benign warnings like this into fatal errors. But that flag keeps the build clean, so sometimes small fixes like this are needed.

Thanks for the fix :)

closed 2023-09-13 by kraiskil