kraiskil/onnx2c #38: installation issue ubuntu - Protocol Buffer version

forgi86, 2023-11-19:
I get an error running:

```
$ make onnx2c
```

The error is triggered by the generated file onnx.pb.h and reads:

"This file was generated by a newer version of protoc which is
incompatible with your Protocol Buffer headers. Please update
 your headers."

Seems related to Protocol Buffer versione, mine is 3.12.4

Marco

kraiskil, 2023-11-20:
This sounds quite strange. Like your `libprotobuf` and the `protoc` are of different versions. Which version of Ubuntu is this?

Does the versions of the following match:
 - `pkg-config --modversion  protobuf`
 - `protoc --version`
 
And what is `GOOGLE_PROTOBUF_VERSION` defined as in the generated `onnx.pb.h` file (in root of the build directory)

 If there is a mismatch between the two, maybe Ubuntu packages several versions of protobuf, and doesn't enforce consistency?

forgi86, 2023-11-21:
``$ lsb_release -a``
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

``pkg-config --modversion  protobuf``
3.12.4

``protoc --version``
libprotoc 3.20.1

There is no ``GOOGLE_PROTOBUF_VERSION`` #define in the file ``onnx.pb.h``

kraiskil, 2023-11-22:
That looks like the problem. I don't know protobuf closely, but I think the `protoc` and `libprotobuf` must be of the same version. onnx2c build system doesn't check for the versions, just takes whatever it finds first.

Can you install a newer libprotobuf or older protoc?

forgi86, 2023-11-23:
OK I found my problem, I was using a conda environment which was masking my /usr/bin/protoc to another one with the wrong version 3.20.1.  A ``$ conda deactivate`` before compilation solved my issue. Thanks for the hint!