kraiskil/onnx2c #39: void* __builtin_memset(void*, int, long unsigned int)’ offset [33, 40] from the object at ‘onnx::_TensorProto_Segment_default_instance_’ is out of the bounds of referenced subobject ‘onnx::TensorProto_Segment::begin_’ with type ‘long int’ at offset 24 [-Werror=array-bounds]    71 |   return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest));

hygxy, 2023-12-01:
```bash
cmake -DCMAKE_BUILD_TYPE=Release .. 
make onnx2c
```
leads to the following error:
```bash
In function ‘void* memset(void*, int, size_t)’,
    inlined from ‘void onnx::TensorProto_Segment::SharedCtor()’ at /home/test/onnx2c/build/onnx.pb.cc:4176:11,
    inlined from ‘onnx::TensorProto_Segment::TensorProto_Segment()’ at /home/test/onnx2c/build/onnx.pb.cc:4161:13,
    inlined from ‘void InitDefaultsscc_info_TensorProto_Segment_onnx_2eproto()’ at /home/test/onnx2c/build/onnx.pb.cc:4159:1:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:71:33: error: ‘void* __builtin_memset(void*, int, long unsigned int)’ offset [33, 40] from the object at ‘onnx::_TensorProto_Segment_default_instance_’ is out of the bounds of referenced subobject ‘onnx::TensorProto_Segment::begin_’ with type ‘long int’ at offset 24 [-Werror=array-bounds]
   71 |   return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
make[3]: *** [CMakeFiles/onnx2c_lib.dir/build.make:167: CMakeFiles/onnx2c_lib.dir/onnx.pb.cc.o] Error 1
make[3]: *** Waiting for unfinished jobs....
[ 50%] Building C object cmake_timestamp/CMakeFiles/timestamp.dir/timestamp.c.o
[ 50%] Built target timestamp
make[2]: *** [CMakeFiles/Makefile2:1228: CMakeFiles/onnx2c_lib.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1262: CMakeFiles/onnx2c.dir/rule] Error 2
make: *** [Makefile:192: onnx2c] Error 2
```

kraiskil, 2023-12-03:
Hmm, Works For Me... This looks like a protobuf problem.
What version of protobuf and whcih compiler are you using?

T-K-233, 2024-01-09:
Same issue here.

```bash
$ protoc --version
libprotoc 3.6.1
```

T-K-233, 2024-01-09:
I wonder if there's a recommended / example way of installing protobuf on Ubuntu 20.04 that I can reference? Thanks!

kraiskil, 2024-01-11:
Installing protobufs should not require any extra tricks. `apt install protobuf-compiler libprotobuf-dev` should be good.

My other development machine is also Ubuntu 20.04, with the same versions of the protobufs, and I don't see this problem - because I always build the `-DCMAKE_BUILD_TYPE=Debug` version...  And this reminded me of:

https://github.com/onnx/onnx/issues/4756 
So to me it looks like a bug in either onnx or protobuf. I'm not sure if anything could be done on onnx2c-side, except maybe check for protobuf versions >3.6 in CMake.

Building the debug version of onnx2c is quick fix for this. But seems it only hides some bug somewhere...

T-K-233, 2024-01-11:
I tried `apt install protobuf-compiler libprotobuf-dev`, which installed protobuf 3.6.1.

With 3.6.1 it still generates the above signal.

**Solution**

I tried build a more recent version of protobuf from source and now it's working for me. Here's what I did:

Install dependencies

```bash
sudo apt install autogen autoconf libtool
```

Clone the source

```bash
git clone https://github.com/protocolbuffers/protobuf.git
cd protobuf
git checkout v3.14.0
git submodule update --init --recursive
```

And then following instructions in the corresponding [README](https://github.com/protocolbuffers/protobuf/blob/v3.14.0/src/README.md)

The command I end up running

```bash
./autogen.sh

./configure --prefix=/usr

make -j8

sudo make install

sudo ldconfig # refresh shared library cache.
```

With the newer protobuf the error is gone.

shubham0204, 2024-01-29:
@kraiskil Building the `Debug` version of `onnx2c` resolves the issue. How about adding this error as a FAQ in `README.md`? Do let me know if I can add it.

kraiskil, 2024-01-29:
@shubham0204 good idea. I added a note in the Readme.

closed 2024-01-29 by kraiskil