kraiskil/onnx2c #43: License question

MatthewScholefield, 2024-03-17:
Hello! This is an awesome project! I'm interested in trying it out but the license language isn't standard even though it is very permissive. I was wondering, would you be open to changing the license (or dual licensing) to something OSI-approved like MIT or BSD 2-Clause?

Warm regards,
Matthew

kraiskil, 2024-03-17:
Hi Matthew.

It is not quite clear to me what those licenses would give. When stripped of their redundancies they boil down to the same as the onnx2c license (+ some small extra restrictions).

What is the problem that needs solving?

Personally I am actually more inclined towards GPL-2, but since I have no practical way of protecting the copyrights of my work, the license question a bit academic. I don't know what the other copyright holders think...

MatthewScholefield, 2024-03-17:
I did a little research since I wasn't fully certain myself and basically from what I understand, I think you get two things:
1. The language of most existing licenses is drafted with very specific "lawyer speak" or so-to-say which means that each word is chosen in a way that reduces ambiguity and provides consistency with other legal documents. Without this, it can cause a legal risk to both the original code authors and to people who use the code. Here are two examples I could think of that apply to the current license:
   - Risk to original code authors: The clause to disclaim liability doesn't do so in a precise manner. A company could suffer serious losses due to a vulnerability in the code and seek compensation claiming that in this case it didn't cause any individual direct harm but rather caused indirect harm which the license didn't expressly disclaim liability for.
   - Risk to code users: The license doesn't explicitly say users may modify the provided source code. If someone does so, they could later be sued by an original code author claiming the original license didn't permit code modification; only free use and unmodified distribution.
3. Due to the potential risks above, many companies (and developers like myself) prefer to only use software that is licensed with a copyleft OSI-approved license. Knowing the license follows specific language streamlines approvals and makes it easy for people to feel safe about using it.

jonnor, 2024-12-10:
Any organization that considers using a software need to check that it has an acceptable license. Using a standard license means that existing license approvals can be done. In that case, approval for a software becomes a trivial operation - check/document that the software uses a pre-existing approved license - done.
For a custom license, like the one used here, a specific evaluation needs to be done, usually involving legal department and/or external counsel. This is very costly. Often this is so costly/tedious that it is not even started. And the software is just considered "not approved" forever. This will likely be a barrier to adoption.

Many have written about this in more detail, the reasons to not create custom licenses for open source software. First example that comes to mind: https://ben.balter.com/2016/08/01/why-you-shouldnt-write-your-own-open-source-license/

kraiskil, 2025-06-08:
What do the fellow copyright holders of onnx2c think of this?
@robinvanemden @YoungsunUConn @AUTOMATIC1111 @tlane-rbx @can-leh-emmtrix @FeKrtz 

The original license was intended as a substitute for 'public domain', which was not legally possible originally.

One "clear" issue with the current license is that it doesn't explicitly allow modification of the sources. This is kind of implied as onnx2c is an open source project. But yes, explicit is better than implicit, and this has been requested via other channels too.

Another easy solution is to re-license under e.g. MIT. It just contains a lot of words, that might mean whatever in lawyer speak. But I guess that is an acceptable risk to take :)

tlane-rbx, 2025-06-09:
Personally I'm ok with the MIT license.

can-leh-emmtrix, 2025-06-11:
Me and my employer would also welcome relicensing my previous contributions (https://github.com/kraiskil/onnx2c/pull/63, https://github.com/kraiskil/onnx2c/pull/64, https://github.com/kraiskil/onnx2c/pull/65, https://github.com/kraiskil/onnx2c/pull/67, https://github.com/kraiskil/onnx2c/pull/73, https://github.com/kraiskil/onnx2c/pull/74) to onnx2c under the MIT license.

FeKrtz, 2025-06-13:
I am also fine with the MIT license 
(#79 )

kraiskil, 2025-07-04:
I opened the PR #83 to switch to the MIT license. Still pending a few approvals.

robinvanemden, 2025-08-12:
My apologies for the late reply! MIT license is perfectly ok to me, too.