kraiskil/onnx2c #48: Error: [Fatal] (createNode) Unimplemented: node operation LinearClassifier

pd-vt, 2024-04-19:
Trying to generate model.c for very Simple classification model which just says input value is positive or negative.

kraiskil, 2024-04-19:
I'm looking at the list of ONNX operators: https://github.com/onnx/onnx/blob/main/docs/Operators.md, and there is no `LinearClassifier`. What is happening? How did you create that model?

pd-vt, 2024-04-20:
The above attached mode is been created using following simple python script. Goal is to just make sure the microcontroller we have is capable to run such basic model and then we go for more complex system. 

`import numpy as np
from sklearn.linear_model import LogisticRegression
from skl2onnx import convert_sklearn
from skl2onnx.common.data_types import FloatTensorType

# Generate some sample data (numbers) and their corresponding labels
numbers = np.array([-2, -1, 0, 1, 2, 3, -3, -4, 5, -5])
labels = np.array(['negative', 'negative', 'positive', 'positive', 'positive', 'positive', 'negative', 'negative', 'positive', 'negative'])

# Reshape the data for compatibility with Scikit-learn
numbers = numbers.reshape(-1, 1)

# Initialize and train a logistic regression model
model = LogisticRegression()
model.fit(numbers, labels)

# Define the input type for the ONNX model
initial_type = [('input', FloatTensorType([None, 1]))]

# Convert the Scikit-learn model to ONNX format
onnx_model = convert_sklearn(model, initial_types=initial_type)

# Save the ONNX model to a file
with open("model.onnx", "wb") as f:
    f.write(onnx_model.SerializeToString())
`

kraiskil, 2024-04-23:
Ah, seems there are two layers of ONNX, the base neural networks and on top of this classical ML operands. https://github.com/onnx/onnx/blob/main/docs/Overview.md

As of now, onnx2c has only considered the NN part of ONNX.

A quick look at ONNX-ML suggests the structure of the opearators are similar in ONNX-ML and ONNX-NN, so implementing ONNX-ML operands might not require (m)any changes to onnx2c internals.
I can't find any detailed info on the ML internals, nor any backend tests for the ML operands, though. But seeing the error in the title of this issue is encouraging - at least onnx2c hadn't crashed on ONNX-ML input...

Adding a new operand to onnx2c isn't too difficult. Much of the work would be to make a few tests for it, as ONNX doesn't provide any. PRs welcome :) https://github.com/kraiskil/onnx2c/blob/master/development.md#adding-a-new-nodeop-implementation

kraiskil, 2026-10-09:
Linear classifier added in #134, thanks @mischirmer

closed 2026-10-09 by kraiskil