kraiskil/onnx2c #50: [Fatal] (createNode) Unimplemented: node operation SplitToSequence

Khubaib96, 2024-05-30:
This is my onnx model.
[denoiser.onnx](https://we.tl/t-thoBbnuyKy)

when i run the command:

`./onnx2c ./denoiser_model.onnx > model.c`

I get these error:

`onnx2c: /mnt/d/onnx2c/src/graph.cc:549: toC::Node* toC::Graph::createNode(std::string): Assertion `false' failed.
Aborted`

How to resolve this?

kraiskil, 2024-05-30:
Hi Khubaib96

> How to resolve this?

These two links should get you going:
https://github.com/kraiskil/onnx2c/blob/master/development.md#adding-a-new-nodeop-implementation
https://github.com/onnx/onnx/blob/main/docs/Operators.md#SplitToSequence

But quickly skimming through the SplitToSequence operator documentation, it does feel like its output tensor's dimension could maybe be data dependent (i.e. determinable only at run-time)? If so, implementing that would require dynamic memory allocation, which onnx2c does not allow.

Khubaib96, 2024-05-31:
Hi, thank you for your quick response.

This model takes 480 input samples and returns 480 output samples. If we provide fewer than 480 input samples, it automatically adds zeros to make it 480 (as a part of preprocessing) not a part of model itself. I'll check my model to see if we can make it work statically.