kraiskil/onnx2c #54: [Fatal] (createNode) Unimplemented: node operation GRU

YadongChen-1016, 2024-09-13:
hi,

I‘m trying to convert a denoiser onnx. Is there no support for GRU conversion yet?

kraiskil, 2024-09-14:
Yup, that is correct. 
After implementing (and bugfixing) LSTM, I didn't have the willpower to redo that for GRU. Also, it is really nice to have an actual network to run inference over to verify the implementation - ONNX unit tests don't offer full coverage.

But with GRU and LSTM being so similar, a version of GRU could be done reasonably quickly by starting with a copy of the LSTM implementation.

kraiskil, 2026-10-09:
GRU added in #137, thanks @mischirmer !

kraiskil, 2026-10-09:
Sorry @Andrew-Infinite I missed your comment. Plenty of stuff left to do. Just do it, submit a PR, and I'll review. 
Don't worry about messing up. If you manage to mess something up, the PR won't be merged in. Just go for it!
Some hints in https://github.com/kraiskil/onnx2c/blob/master/development.md

closed 2026-10-09 by kraiskil