kraiskil/onnx2c #72: onnx2c ( main branch)  does not link with protobuf 26.2

gabessolo, 2025-04-05:
Hello

[image]
[image]

,

Thank you very much for your time and support.
I have checked the version of protobuf and protoc and it is the same (26.2)
but i have got the same issue when linking onnx2c with protobuf.
( a screenshot of the build is included)
*********
wwwdata@serveur:~/downloads/onnx2c/build$ **lsb_release -a**
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy
wwwdata@serveur:~/downloads/onnx2c/build$ **pkg-config --modversion  protobuf
26.2.0**
wwwdata@serveur:~/downloads/onnx2c/build$ **protoc --version**
libprotoc 26.2-dev
wwwdata@serveur:~/downloads/onnx2c/build$ cmake --build .
Consolidate compiler generated dependencies of target onnx2c_lib
[  1%] Built target onnx2c_lib
*********
CMake Warning at /usr/share/cmake-3.22/Modules/FindProtobuf.cmake:524 (message):
  Protobuf compiler version 26.2 doesn't match library version 5.26.2
Call Stack (most recent call first):
  CMakeLists.txt:18 (find_package)
*************************
-- Generating done
-- Build files have been written to: /home/wwwdata/downloads/onnx2c/build
[  1%] Built target clear_cache
Consolidate compiler generated dependencies of target timestamp
[  1%] Building C object cmake_timestamp/CMakeFiles/timestamp.dir/timestamp.c.o
[  1%] Built target timestamp
Consolidate compiler generated dependencies of target onnx2c
[  1%] Linking CXX executable onnx2c
/usr/bin/ld: libonnx2c_lib.a(onnx.pb.cc.o): warning: relocation against `_ZN6google8protobuf8internal26fixed_address_empty_stringB5cxx11E' in read-only section `.text.startup'
/usr/bin/ld: CMakeFiles/onnx2c.dir/src/main.cc.o: in function `main':
main.cc:(.text.startup+0xab): undefined reference to `google::protobuf::MessageLite::ParseFromIstream(std::istream*)'
/usr/bin/ld: libonnx2c_lib.a(graph.cc.o): in function `toC::Graph::getIoTensor(onnx::ValueInfoProto&)':
graph.cc:(.text+0x10b46): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
/usr/bin/ld: graph.cc:(.text+0x10b5d): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
/usr/bin/ld: graph.cc:(.text+0x10f04): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
/usr/bin/ld: graph.cc:(.text+0x11214): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
/usr/bin/ld: graph.cc:(.text+0x1191f): undefined reference to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]'
/usr/bin/ld: libonnx2c_lib.a(onnx.pb.cc.o): in function `onnx::TensorShapeProto_Dimension::Clear()':
onnx.pb.cc:(.text+0x265): undefined reference to `google::protobuf::internal::ArenaStringPtr::Destroy()'
*******************************************
........
.......

gabessolo, 2025-04-05:
**THE BEST FIT IS protobuf  3.12.4 and protoc 3.12.4**
*************************************

**make onnx2c**

-- Failed to find LLVM FileCheck
-- git version: v1.7.0-10-g77d1e74d normalized to 1.7.0.10
-- Version: 1.7.0.10
-- Performing Test HAVE_STD_REGEX -- success
-- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
-- Performing Test HAVE_POSIX_REGEX -- success
-- Performing Test HAVE_STEADY_CLOCK -- success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wwwdata/downloads/onnx2c/build
[  0%] Built target clear_cache
Consolidate compiler generated dependencies of target timestamp
[  0%] Building C object cmake_timestamp/CMakeFiles/timestamp.dir/timestamp.c.o
[  0%] Built target timestamp
Consolidate compiler generated dependencies of target onnx2c_lib
[100%] Built target onnx2c_lib
Consolidate compiler generated dependencies of target onnx2c
[100%] Linking CXX executable onnx2c
[100%] Built target onnx2c

*************************************
**pkg-config --modversion protobuf**
3.12.4
**protoc --version**
libprotoc 3.12.4
 **lsb_release -a**
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy

closed 2025-04-05 by gabessolo

kraiskil, 2025-04-14:
Thanks for the update. Happy you got it solved.