kraiskil/onnx2c #76: entry( tensor_image , tensor_grid)  how to interpret in term of picture tensor_grid

gabessolo, 2025-04-10:
Hi,

I want to use entry(tensor_image, tensor_grid)  to detect object and classify and also.
How to achieve this goal ?

This is my code below:
void entry(const float tensor_image[1][3][416][416], float tensor_grid[1][125][13][13]);

int width = 0, height = 0, bpp = 0;
float tensor_image[1][3][416][416],tensor_grid[1][125][13][13];

int main() {
		
    uint8_t* rgb_image = stbi_load("input.png", &width, &height, &bpp, 3);
	
	width  = width  <= 416 ? width  : 416;
	height = height <= 416 ? height : 416;
	
	uint8_t* pScanRgbImage =  rgb_image;
	
	int x = 0, y = 0 , z = 0;
	
	for (int i = 0; i < 416 * 416; ++i) {
		
	  tensor_image[0][0][x][y] = 0;
	  tensor_image[0][1][x][y] = 0;
	  tensor_image[0][2][x][y] = 0;
	  
	  x = (x == 416 ? 0 : x );
	  
	  x++;
	  
	  y = (x == 416 ? y++ : y );
	  y = (y == 416 ? 0 : y );
	 
	}
	
	
	for (int i = 0; i < width * height; ++i) {
		
	  uint8_t red   = *pScanRgbImage++;
	  uint8_t green = *pScanRgbImage++;
	  uint8_t blue  = *pScanRgbImage++;
	  uint8_t alpha = *pScanRgbImage++;
	  
	  tensor_image[0][0][x][y] = (float)red;
	  tensor_image[0][1][x][y] = (float)green;
	  tensor_image[0][2][x][y] = (float)blue;
	  
	  x = (x == 416 ? 0 : x );
	  
	  x++;
	  
	  y = (x == 416 ? y++ : y );
	  y = (y == 416 ? 0 : y );
	  
	}
	
	stbi_image_free(rgb_image);
	
	entry(tensor_image,tensor_grid); // INFERENCE 
	
    memcpy(rgb_image,0,416*416);
	
    rgb_image = malloc(width*height*CHANNEL_NUM);

	// Write your code to populate rgb_image here
	for (int i = 0; i < 13 * 13; ++i) {
		
	  tensor_grid[0][0][x][y] ;
	  tensor_grid[0][1][x][y] ;
	  tensor_grid[0][2][x][y] ;
	 
	}
    
	stbi_write_png("output.png", width, height, CHANNEL_NUM, rgb_image, width*CHANNEL_NUM);
	
	//int width = 800;
	//int height = 800;

	// resize

	stbi_image_free(rgb_image);

    return 0;
}

kraiskil, 2025-04-14:
How the input and output data should be laid out and interpreted is something that is explained with the documentation of the NN model. Onnx2c only compiles the .onnx file to C. The dimensions and their meaning in the input and output tensors match in the C code and the model/python code.

Sometimes such documentation is not explicit, but buried in some example python script shipped with the model and left for the reader to figure out.

gabessolo, 2025-04-18:
those links below  are a very good starting point for me to achieve the interpretation of tensor grid generated
by onnx2c, so i have achieve my project. Thanks a lot for your guidelines and your help.

If you know any companies that want to integrate yolo into their project, I could sell them my code.
And if you put me in touch with them and they buy, I can pay you up to 20%.
 
my code implement the interpretation of the output of the entry function generated by onnx2c 
it is written in modern C++ and can be build for different target.

Regards

LINKS :

[https://pjreddie.com/darknet/yolo/](https://www.linkedin.com/safety/go?url=https%3A%2F%2Fpjreddie.com%2Fdarknet%2Fyolo%2F&trk=flagship-messaging-web&messageThreadUrn=urn%3Ali%3AmessagingThread%3A2-NTc0M2FjZGItNzg0My00NDg2LWEzN2QtYzJkYzVkZjcxNzkzXzEwMA%3D%3D)

[https://learn.microsoft.com/fr-fr/dotnet/machine-learning/tutorials/object-detection-onnx](https://www.linkedin.com/safety/go?url=https%3A%2F%2Flearn.microsoft.com%2Ffr-fr%2Fdotnet%2Fmachine-learning%2Ftutorials%2Fobject-detection-onnx&trk=flagship-messaging-web&messageThreadUrn=urn%3Ali%3AmessagingThread%3A2-NTc0M2FjZGItNzg0My00NDg2LWEzN2QtYzJkYzVkZjcxNzkzXzEwMA%3D%3D)

gabessolo, 2025-04-23:
[image]
[image]
[image]

gabessolo, 2025-04-23:
Hello,
I have successfully
build protobuf
build onnx2c
generate tinyyolov2-7.c
build libtinyyolov2-7.a
build the executable ia_detect based on the lib libtinyyolov2-7.a

now i need to connect to people looking for consultancy, source code on the topic of yolo, computer vision
if you help i will give you 20% of each deal

kraiskil, 2025-04-26:
Nice that it started working.

closed 2025-04-26 by kraiskil