kraiskil/onnx2c #82: Compilation error "ERROR" redefined

HerGeo, 2025-06-26:
I tried to compile onnx2c but am getting the following error:
```
In file included from C:\Arbeit\ONNX2C\onnx2c\src\graph.cc:2:
C:\Arbeit\ONNX2C\onnx2c\src\error.h:4: error: "ERROR" redefined [-Werror]
    4 | #define ERROR(why) do { \
      |
compilation terminated due to -Wfatal-errors.
cc1plus.exe: all warnings being treated as errors
```
What am I doing wrong?

HerGeo, 2025-06-26:
It seems to be a Windows related problem. Compiling in Ubuntu run completly fine.
I tried adding an include_guard to the error.h file, but without effect. However, commenting out the define statement does the trick.
Unfortunately, this brings me to the next error (which is raising on Windows only too).

Could someone give a hint, where this platform related issues origin from? Maybe dependencies, that are platform specific or diffenreces in in compiler behaviour?

kraiskil, 2025-07-04:
Hi HerGeo

yeah, the file is missing include guards, that at leas is wrong. Being 100% on Linux myself, I didn't bump into this.

But if adding the include guards doesn't solve this, then something else in the Windows toolchain already defines the preprocessor macro `ERROR`. Maybe even aixlog itself. 

To hunt it down you can add lines like:
```
#ifdef ERROR
#error ERROR already defined
#endif 
```
Before & between the `#include`s to track down where the error gets defined.

Or maybe it is Windows ignoring case, and interpreting `#ERROR` as the keyword `#error`?

The smart thing anyway would have been to name the macro `ONNX2C_ERROR`...